espotek-org/Labrador #126: Enable SUMP or other Sigrok compatible protocol

khimaros, 2020-07-29:
Sigrok is an open source, cross platform, logic analyzer frontend with support for higher level protocol decoders such as I2C, SPI, UART, etc. By speaking a protocol Sigrok speaks, the Labrador could tap into that potential.

EspoTek, 2020-07-30:
This is something I looked into a couple of years ago.  Unfortunately, due to some early design decisions (some which I stick by, some which I regret), it's extremely difficult to translate Labrador's data streams into a form that Sigrok will understand.

Give us 5 years and I'm sure there'll be a Sigrok-compatible Labrador model 2.0, though!

closed 2020-07-30 by EspoTek

khimaros, 2020-07-30:
Thank you for the reply @EspoTek. Is this a limitation of the hardware or just a limitation of the current firmware implementation?

EspoTek, 2020-07-30:
Both, actually.  The board sends raw samples over a USB isochronous stream to offload the heavy lifting to the PC-side software (that's the decision I stick by), and the current code for sampling and decoding the stream is not properly modularised (that's the decision I regret).