espotek-org/Labrador #127: Is the samplerate dependent on CLKDIV ?

jhegedus42, 2020-08-04:
Dear All, 

I am reading this : 

[image]

From here : https://docs.google.com/document/d/1ZDO1RTarQTNB6Pdfi_T4YukL51oYan_kk_teb1cu6-o/

I am not sure if changing `PER` or `CLKDIV` changes the samplerate.

Does it ?

Or, how can I make sure that the sample rate is what I believe it is (375kbps) ?

Does it change ? Somehow ?

I mean, I see that in librador I can set the samplerate. But does that actually 
change the sample rate of the device ? Or does it just do a resampling + 
interpolation ? Here it seems to me it does "simple" resampling : https://github.com/EspoTek/Labrador/blob/d71b72fdabb3cf2a3f77c2df468b9e405adac846/Librador_API/___librador/librador_shared_library/librador.cpp#L64 "in software". So the Labrador's sample rate is either 375kbps or 750 kbps, is
that right ?

Does the `PER` or `CLKDIV`  change the actual sample rate ? I would think not, but asking it just to be sure.

Kind Regards,

Jozsef

EspoTek, 2020-08-04:
PER and CLKDIV control the signal generator.
You'll want to use the mode control to set oscillscope mode.
https://docs.google.com/document/d/1ZDO1RTarQTNB6Pdfi_T4YukL51oYan_kk_teb1cu6-o/edit#heading=h.4y073861btqr

closed 2020-08-04 by EspoTek

jhegedus42, 2020-08-04:
Thanks ! I look into that ! Cheers, Jozsef