espotek-org/Labrador #129: Android 8 device not connected

doc-aka-rich, 2020-10-01:
Device is Samsung Galaxy S7 Edge
Android is verizon build of Android 8.0
OTG adapter tested connecting a random selected USB flash ... phone can read directory
  (pre-answering other posted issues questioning whether OTG adapter connects data pins)

Connecting Labrador board to phone thru OTG adapter automatically launches Labrador app

Near the top of the app it reads in red text "device not connected"

I read another closed issue suggesting disconnecting and reconnecting Labrador board ... did that multiple times ... red text doesn't go away

I also read in a closed issue that the board may be connected regardless of the red text.  I selected CALIBRATE, and it says not without a board connected.

EspoTek, 2020-10-03:
Does the board work on Desktop?  Officially, the app is only tested up to Android 7.1 but user reports suggest it should work on 8.0.

skorokithakis, 2020-12-08:
It works for me on Android 10, but needs calibration every time it starts.

doc-aka-rich, 2020-12-08:
A new reply reminded me I left this issue "open".

I'm marking it "closed".  I tossed the board onto a shelf, as whether it works on a desktop / windows PC is of no interest or value ... I have working USB oscilloscopes that work on both windows and linux.  The only interest of this board was the hope of it working on android phone.  My phone is android 8, and I've heard no remedy or diagnostic.

I consider it "closed", and am marking it so on GitHub.

closed 2020-12-08 by doc-aka-rich

skorokithakis, 2020-12-08:
@doc-aka-rich which USB oscilloscopes do you have?