espotek-org/Labrador #130: Labrador crashes on MacOS 10.11.6

rickspratt, 2020-10-03:
I just downloaded and opened the .dmg file and double clicked on the Labrador app.  It briefly flashes in the tool bar below and then goes away.  I have MacOS 10.11.6 installed on my MacBook Pro.  This happens whether or not the EspoTek board is plugged into the USB port.  I couldn't even load it on my MacMini running MacOS 10.15.5 because it isn't a verified Mac App.  I would hate to think I just threw away $30 on this.

EspoTek, 2020-10-03:
Yep, that's correct.  The app is unsigned.  You'll need to open the Security page under the Settings app after attempting to launch Labrador and manually allow it.  This is common for many open-source and free apps.

tatianaper, 2020-10-05:
I am having a similar issue. I have already allowed it on my security setting, but it is still flashing and then disappearing.

EspoTek, 2020-10-05:
If you run the app through Terminal (right click on .app bundle, show contents, then navigate to Contents->MacOS->Labrador), what does it say?

tatianaper, 2020-10-05:
Last login: Mon Oct  5 18:18:03 on ttys001
Tatiana-Pereiras-MacBook-Pro:~ tatianapereira$ /Volumes/bin\:Labrador/Labrador.app/Contents/MacOS/Labrador ; exit;
Sorry, "Labrador" cannot be run on this version of macOS. Qt requires macOS 10.12.0 or later, you have macOS 10.10.5.
logout

[Process completed]

So it says I need macOS 10.12.0 or later even though it says for macOS 10.10 and later on the github where I downloaded it. Is there a way for me to get a version that is compatible with my computer?

EspoTek, 2020-10-05:
Thanks, Tatiana.  That was a clue I needed!
It seems I accidentally broke compatibility when upgrading the Qt version late last year.  I'll get working on a patch now.

EspoTek, 2020-10-05:
There we are, that was easy.
I've just compiled a version with Qt 5.9.
Could I please ask you to confirm that this is working as I don't have a machine running Yosemite.
https://github.com/EspoTek/Labrador/releases/tag/continuous-macOS

tatianaper, 2020-10-05:
This link sends me to the same page I was using before. I tried again anyway and it is still giving me the same problem. I get the same message that I need OS 10.12 despite the github page stating otherwise.

EspoTek, 2020-10-06:
I'll get a VM up and running and fix this ASAP.

tatianaper, 2020-10-06:
Great! I look forward to hearing from you.

rickspratt, 2020-10-06:
I tried running it from a terminal using the latest release on MacOS 10.11.6 and got the same error message: Sorry, "Labrador" cannot be run on this version of macOS. Qt requires macOS 10.12.0 or later, you have macOS 10.11.6.

tatianaper, 2020-10-06:
I have macOS 10.10.5

EspoTek, 2020-10-09:
No luck in the VM, but there is a version that's compatible with older versions of OSX:
https://github.com/EspoTek/Labrador/releases/tag/1.1

Going to mark this issue as solved for now, if you absolutely must have the latest version of the software then I recommend either building from source (which involves a maze of dead links) or upgrading to Sierra.

closed 2020-10-09 by EspoTek