espotek-org/Labrador #131: Incompatible with OSX 11.0 Big Sur

EspoTek, 2020-11-23:
Crashes when rendering first QMenu.

Current workaround is to enter the command `export QT_MAC_WANTS_LAYER=1` before launching from terminal.
Upgrading to Qt 5.15 *should* fix this?

itopaloglu83, 2021-02-17:
Here's a quick work around for anyone who doesn't want to keep using the Terminal app each time.

Open the terminal app and navigate to the app directory.

```
cd /Applications/Labrador.app/Contents/MacOS
```

Rename the original executable to a different name.

```
mv Labrador FixLabrador
```

Create a new shell script with original executable name of `Labrador.sh`.

```
#!/bin/zsh
export QT_MAC_WANTS_LAYER=1
/Applications/Labrador.app/Contents/MacOS/FixLabrador
```

Update file permissions to for the new shell script.

```
chmod +x Labrador.sh
```

You should now be able to launch the app directly from its icon without any issues. It's still a workaround though.

itopaloglu83, 2021-02-17:
Never mind, there's a better workaround from another repo.

https://github.com/spyder-ide/spyder/issues/14222#issuecomment-729493687

itopaloglu83, 2021-02-17:
@EspoTek, could you add the following lines into `Desktop_Interface/resources/Info.plist` to fix this issue?

```
<key>LSEnvironment</key>
<dict>
    <key>QT_MAC_WANTS_LAYER</key>
    <string>1</string>
</dict>
```

EspoTek, 2021-02-17:
That's amazing.  I had no idea you should do things like that!
Thanks so much, Ihsan.

I wish it were as simple as just updating the .plist post-build.  Unfortunately, I rely on Travis and need to move build systems before I can release a new OSX/Linux version.  I'll look into it over the coming weeks, as this issue seems to be coming up more and more often now.

EspoTek, 2021-03-05:
This bug should be fixed in the following release:
https://github.com/EspoTek/Labrador/releases/tag/continuous-macOS

Would someone kindly be able to confirm this?

itopaloglu83, 2021-03-05:
I was able to launch the app without any issues. `macOS Big Sur Version 11.2.2`

[image]

EspoTek, 2021-03-05:
Fantastic!  Marking this as closed now.

closed 2021-03-05 by EspoTek