espotek-org/Labrador #132: Feature Request: Sigrok integration

emosenkis, 2020-12-02:
It would be amazing to be able to capture data directly into [Sigrok](https://sigrok.org/). I've managed to dump it to CSV then load it in Sigrok, but I think Labrador could really benefit from Sigrok's mature [PulseView](https://sigrok.org/wiki/PulseView) UI and analysis capabilities.

EspoTek, 2020-12-03:
I completely agree with you that it would add a huge amount of value to Labrador, especially for the logic analyser part.
Unfortunately, I made some bad upfront design decisions that make Sigrok integration far from trivial.
Perhaps in Labrador 2.0...

closed 2020-12-03 by EspoTek

techdragon, 2021-12-07:
@EspoTek Could you outline some of these design issues? ... I saw an old post about Librador and an API in reference to potential Sigrok support and I see old code for it here still... I adore the idea of having a few cheap 12bit Oscilloscopes for when I need that little extra, and it would be utterly wonderful to just be able to throw everything into Sigrok with my other scopes and such.

nerdx86, 2022-01-05:
@EspoTek @techdragon Was there any further conversation on this?  I or my team may be able to throw some help this way as well?

EspoTek, 2022-01-06:
Nah mate, nothing to add.

You're welcome to have a crack, but this would be a serious undertaking!