espotek-org/Labrador #133: Calibration Issues

wtpearson, 2020-12-11:
Hello, I'm having issues getting my new board calibrated.  Channel 2 has a "permanent" +1.7v offset and is noisier than channel 1. Grounding both channels to the USB shield (or any other ground) helps with the noise a bit but I still can't calibrate due to the offset. 
Not grounded:
[image]

Grounded:
[image]

Thanks.

EspoTek, 2020-12-12:
Hi mate.  The channels should float to around 1.7V when no input is connected.
Are you absolutely sure you're grounding the correct pins?
Make sure you're using the DC inputs, *not the AC*, and double check you're familiar with the pinout.
https://github.com/EspoTek/Labrador/wiki/Pinout

Top row is CH2, bottom is CH1.

wtpearson, 2020-12-14:
Oof, yes I should have examined the pinout more carefully.  I was using the incorrect channels for calibration.  Thanks for the assist.

closed 2020-12-14 by wtpearson