espotek-org/Labrador #134: Windows 7 drivers not found after install

Ikemeister, 2020-12-11:
Can't get this unit to work with my Pixel 3a XL so thought I'd try it out on my Windows 7 PC. 
[image]
Ran the installation in spite of the warning. The install proceeds to the Prerequisites setup screen:
[image]
I left them all selected:
[image]
Bootloader installed:
[image]
Primary Installed:
[image]
Misconfigured Batch Installed:
[image]

I continued with the installation but the device shows up as Not Connected:
[image]

It shows up as unknown via Device Manager:
[image]
Searched for the drivers using the documented procedure. Note that the drivers aren't found: 
[image]

I repeated the driver installation several times with same results. So what now? Could I perhaps explore for the drivers where they should have been installed?  Suggestions please!

EspoTek, 2020-12-12:
Hi mate.  Under the "Details" tab, what is the value of the "Hardware IDs" property?

Ikemeister, 2020-12-12:
*ANVSOFTWaveExtensible

EspoTek, 2020-12-12:
That doesn't seem right.  When you unplug the Labrador, does the Unknown Device disappear?

Ikemeister, 2020-12-12:
Nope. Still there. Should the Labrador driver disappear when the unit is disconnected?

Ikemeister, 2020-12-12:
I checked to see what shows up when plugging the unit back in and I find another unknown device under UBC controllers. When I follow the procedure as described in the documentation, I find only Unknown Device is found:
[image]

EspoTek, 2020-12-13:
Yeah, the device should not be visible under Device Manager when it's not connected.  The USB controller is a red herring, too.

When connecting, does the "USB Connected" sound play?
If not, please try the cable you have with a different device and make sure it can transmit data.
If it's not the cable, your board is faulty.

Ikemeister, 2020-12-14:
You are correct Sir! The cable I was using and had been using for a long time, can now only be used for charging purposes. It had been working to transfer data with my PC from my old phone but with the Pixel I've been using WiFi. In the meantime one of the data leads must have broken. It's now working with another USB cable. Apologies and thanks for your suggestions.

closed 2020-12-14 by Ikemeister

EspoTek, 2020-12-14:
No dramas, mate.  Thanks for the detailed information and actually following the suggestions!  Made solving the problem so much easier and more pleasant!