espotek-org/Labrador #136: Simple circuit

ralphxyz, 2020-12-18:
"The top yellow wire connects Vin to both the signal generator CH1 as well as the oscilloscope CH2 pin."

What? How? The picture is very fussy. 

Thanks,
Ralph

FlaryMBT, 2020-12-19:
Ralph,

The lead of R1 closest to the Labrador in the picture (upper lead) is on the same row of the breadboard as the Signal Gen CH1 (AC) pin of the Labrador. Also on this row is the upper yellow wire connected from the breadboard to Oscilloscope CH2 (DC) socket.

Pay careful attention that the lower lead of R1 is connected to the upper lead of R2 and Oscilloscope CH1 (DC) socket of the Labrador through the lower yellow wire and breadboard. GND of the circuit is connected to the Power Supply (GND) pin of the Labrador through the breadboard as well.

M

ralphxyz, 2020-12-21:
Thanks for the reply. It's a shame none of this is disclosed in the
documentation.

Now how do I setup/use the oscilloscope?

Ralph

closed 2021-02-10 by EspoTek