espotek-org/Labrador #137: Win 10; Stuck On Flashing Device Firmware In Software

FlaryMBT, 2020-12-20:
I am trying to setup the Labrador to work on my newish laptop running Windows 10 (64bit).

When I plug in the Labrador, Windows recognizes it as a Labrador under Espotek Devices in Device Manager. So I know that there is nothing wrong with the USB connection.
I then launch the Labrador Software and am met with the Labrador reconnecting as an Atmel Device, and the software saying "Flashing Device Firmware". I open the console to see the following:
>FLASHING  labrafirm_0006_01.hex
>deviceConnectedDisplay::flashingFirmware
>UsbK_ControlTransfer failed with error code 31
>stdio_stage1 ""
>sterr_stage1 "dfu-programmer: no device present. \r\n"
>EXIT_CODE stage1 3
>stdio_stage1 ""
>sterr_stage1 "dfu-programmer: no device present. \r\n"
>EXIT_CODE stage1 3

After unplugging and replugging in the Labrador then waiting 10 seconds, the same thing occurs.
I go through the troubleshooting wizard under Help and it states this:
>It seems like your board is already connected and configured correctly.
>
>If your board is not functioning correctly, this indicates that there is an issue with the USB driver.
>
>Let's go through some manual troubleshooting steps.

I then go through the rest of the troubleshooting wizard and the process does not change anything.
I have also tried entering the Labrador into Bootloader Mode and running the batch files Wipe then Flash. Nothing happens with these and I still run into the same error with the software after reconnecting and taking the Labrador out of Bootloader Mode.

I have tried running the Labrador on my RPi 3B+ and it proceeds through this flawlessly but does not make it past calibration. On the rare chance that the oscilloscope calibrates correctly, I cannot get all the way through calibrating the power supply without getting an error in the console and having to replug in the Labrador.

What am I doing wrong if anything and how can get past this issue?
Thank you in advance,
M

EspoTek, 2020-12-27:
I've only ever seen that happen with one other board - it's a freak issue caused by an edge case bug in dfu-programmer that no-one understands.  Until this bug is fixed, the only solution is to either flash it manually with a JTAG/PDI programmer or swap the board out.
Where did you purchase the board from?

FlaryMBT, 2020-12-28:
Dear Chris,
Thank you for the reply, I had no idea about this issue.
I purchased the Labrador on November 26th, 2020 on Amazon. It was shipped from them and sold by Espotek.

EspoTek, 2020-12-28:
No worries.  Please send an email to admin@espotek.com with your details and I'll get a replacement board sent out to you.

closed 2020-12-28 by EspoTek

Alelanlo, 2022-09-30:
Nevermind. I found another thread with a similar issue and the problem was solved with this:
[image]