espotek-org/Labrador #140: Thought about making a more robust device/version 2?

cweagans, 2021-02-02:
The Labrador is great. It's really a huge accomplishment to have gotten it out the door, especially with how much functionality is packed into such a tiny space!

Have you thought about making version 2? And/or have you thought about making v2 a bigger/more full featured all-in-one device? I've been noodling on this and I think there's a pretty good market in the $100-200 range for a combination power supply/signal generator/oscope/logic analyzer. Here's my wishlist in the event that you're thinking about going down this road:

* Include a case with real ports for the various functions -- current device is good for breadboards and the like, but it would be cool to turn this into a bench tool. Display isn't needed, but having real ports for probes, PSU output, etc. would be cool!
* Bigger -- the current device is very portable, but is also easily lost in e.g. a pile of papers or whatever. Having a bigger, more substantial device would be awesome.
* More channels/higher data rate/etc -- generally just "more" across the board as it relates to the oscope/logic analyzer. I realize this wasn't ever meant to be a top-of-the-line device, but I'm curious about how far in that direction something like this could go with a bit more headroom in the budget? What's a sane number of channels/sample rate/etc?
* Same software -- the desktop client is really, really wonderful and I'd love to keep using it
* Fully variable PSU output - 0-60V up to 100W (which would allow for powering any PoE device, I think). Bonus points for making it possible to do more than one output for powering more complicated devices?
* USB-C -- I threw out all of my micro USB cables several years ago and now only have one for the Labrador. It would be awesome to standardize on USB-C across the board (and might help to accommodate higher sample rates and such too?)

That's most of my list, I think. If you're thinking about doing something like this, I'd love to collab on the software bits. Hardware isn't my forte just yet, but I'm learning.

EspoTek, 2022-01-24:
The short answer is "yes, eventually".

Although I'd take the current software out the back and put a bolt through its skull before updating it for future hardware.  There are a lot of really short-sighted design choices that worked, but make it difficult to maintain long term.

closed 2022-01-24 by EspoTek

cweagans, 2022-01-24:
Fair enough! If you want help with the software bits, definitely hit me up. I’ve been very, very happy with Flutter and Dart if you’re looking for an alternative to the current setup.

EspoTek, 2022-01-24:
The plan was actually to use Dear ImGui.
I'm not sure if you've used it before, but with a little bit of wrapper code to handle windowing and stuff cross-platform, it makes almost every class of GUI-related problem disappear.