espotek-org/Labrador #141: fresh install of EspoTek_Labrador-ac5052c-x64.deb does not run

wfdudley, 2021-02-08:
Ubuntu 16.04
uname -a
Linux django 4.4.0-186-generic #216-Ubuntu SMP Wed Jul 1 05:34:05 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

I installed the .deb file.  Running the application yields this:

/usr/bin/EspoTek-Labrador/Labrador

/usr/bin/EspoTek-Labrador/Labrador: /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5: version `Qt_5' not found (required by /usr/bin/EspoTek-Labrador/Labrador)

/usr/bin/EspoTek-Labrador/Labrador: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/bin/EspoTek-Labrador/Labrador)

/usr/bin/EspoTek-Labrador/Labrador: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5' not found (required by /usr/bin/EspoTek-Labrador/Labrador)

/usr/bin/EspoTek-Labrador/Labrador: /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5: version `Qt_5' not found (required by /usr/bin/EspoTek-Labrador/Labrador)

/usr/bin/EspoTek-Labrador/Labrador: /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5: version `Qt_5' not found (required by /usr/bin/EspoTek-Labrador/Labrador)

EspoTek, 2021-02-08:
Please use the .AppImage or `make install`.
The .deb is no longer officially supported.

closed 2021-02-08 by EspoTek

wfdudley, 2021-02-08:
Perhaps consider updating the web page to state that the deb doesn't work,
so as to save everybody's time.

Thanks,
Bill Dudley

This email is free of malware because I run Linux.

EspoTek, 2021-02-09:
Already done.  Sorry about before!

wfdudley, 2021-02-09:
I realized that the problem is that different versions of Ubuntu LTS are
"stuck" at different Qt library version.
My 16.04 is stuck at 5.5-something, and 18.04 probably has 5.9-something
that your program is
compiled for.

I've successfully used the Appimage version on a 64 bit Linux  (Devuan,
which is debian based) and it works fine.
It won't load on a Linux built using 686 binaries, however.  I'm reluctant
to try building it from source as that
always seems to turn into a huge pageant.

Thanks for your work.
This email is free of malware because I run Linux.

EspoTek, 2021-02-09:
Yeah.  To be honest, that's why I'm going AppImage/Make Install-only from now on.  There are just too many different distributions of Linux in common use, each with their own slight differences and package managers.
Supporting them all through the "proper" methods is just not feasible for an organisation of EspoTek's size (1 guy working part-time!).