espotek-org/Labrador #142: Device shuts off once a simple circuit is formed

maith1, 2021-02-09:
We are using the Labrador for a class electric circuits lab. Our instructions were to create a simple circuit on a breadboard with two wires and a resistor. After doing this, the Labrador shuts off and then wants to recalibrate again. This basically forces me into an infinite loop of recalibration. Is my Labrador broken or could I be doing something wrong?

EspoTek, 2021-02-09:
Which OS are you using?  The calibration routine is completely managed by the PC and has nothing to do with the hardware.

maith1, 2021-02-09:
I am using Windows 10. The calibration seems to go just fine before I plug anything in but once I form a full circuit, the red light on the device shuts off. When I unplug it and plug it back in, it wants to calibrate again.

EspoTek, 2021-02-09:
OK, so it looks like there's two independent issues here.
The first issue is a minor software bug - it seems that asking it to calibrate when the board is connected (on initial launch) will ask it to calibrate _every_ time the board is connected.  Relaunching the software will fix this.

The second issue is most likely caused by an short somewhere in your circuit.  Please check it very, very carefully, and make sure you're aware how the breadboard is laid out.

maith1, 2021-02-10:
I have installed the software on another computer entirely and am still encountering the same problem. My circuit appears to be constructed more or less the same as the one our lab tutorial shows as well. I'm not sure what could be causing this other than maybe a faulty Labrador?

EspoTek, 2021-02-10:
If the light turns off, it means the OCP is tripping.  Either the factory installed a bad fuse (which can be routed around), or there is an issue with the circuit.  Please send me some pics of your attached circuit and I can map out where the short is for you.

maith1, 2021-02-10:
[image]
I am relatively new to doing this kind of stuff so it is highly likely that I am doing something incorrectly. I just set this up the way that my lab instructions appeared to show. I'm sorry if the problem is obvious and I'm missing it.

EspoTek, 2021-02-10:
OK, I can see what's going on here.
Honestly, the problem is obvious to a trained engineer but everyone (myself included) makes this exact kind of error when starting out.

Check out the chart here:
https://www.electronics-tutorials.ws/resistor/res_2.html

The questions to ask yourself are the following:
 - What is the resistance value of the resistor?
 - How much current is it drawing?
 - How much power is being dissipated by it?
 - How does this value compare to the max power that Labrador can supply?

(Obviously it's a leading question, but I recommend doing the maths anyway.  The magnitude differences are pretty impressive!)

maith1, 2021-02-10:
Thank you for the helpful information. I will try to work this out now that I know where to start. I believe now that I can close this ticket. Thank you once again! Sorry for the inconvenience.

EspoTek, 2021-02-10:
I think it's best to leave the ticket open.  While the disconnect wasn't a bug in Labrador, the re-calibration over and over again definitely was!

maith1, 2021-02-10:
Okay! I'm not too familiar with the GitHub ticket system so I'm not really sure of the implications of opening and closing tickets but if you believe it should stay open then I trust your judgment. If I have any more problems then I guess I'll put them here. Thanks again.

EspoTek, 2021-06-24:
Abandoning this issue.  It's low priority and honestly a mess of a system to fix.

closed 2021-06-24 by EspoTek