espotek-org/Labrador #144: Device No longer Connects

mff47025, 2021-02-11:
A couple of weeks ago my board was working fine.  However, over that last week it has stopped connecting.  I have tried reinstalling the software a couple of times.  When I look in the device manager that cable shows up as a libusb-win32 usb device: ATxmega32A4U.
I have tried to change the driver of this device and it doesn't show libusbK devices as an option.  I don't see any other usb devices in my device manager.  I'm running Windows 10.
Open to any ideas, thanks.

EspoTek, 2021-02-11:
There's a Firmware Recovery tool under the "Help" menu that can fix most of these problems.
Specifically on Windows, though, there are a couple of batch files under `C:\Program Files\EspoTek\EspoTek Labrador\firmware`: `flash.bat` and `wipe.bat`.

You'll know it's successful if `flash.bat` hangs around on screen for about a second and prints a message with a whole lot of `>` characters.
On reconnect, the board should come up in device manage correctly.

mff47025, 2021-02-13:
Thanks.  That was it.

closed 2021-02-13 by mff47025