espotek-org/Labrador #145: Device not connected and other issues

hchaudh2, 2021-02-12:
I have tried to download Mac OS Continuous Build from this link: https://github.com/EspoTek/Labrador/releases
however, I am unable to open Labrador application. So I downloaded Mac OSX from this link https://github.com/EspoTek/Labrador/releases/tag/1.1  and it opens however when I try plugging in usb to calibrate, it says device not connected. I have even tried firmware recovery and multiple USB cables especially those that support data transfer. I'm using an adapter so I only have one USB port on it so one device is connected at a time. But even then it says device is not connected. I'm using this for an electric circuits class in college and have a deadline for a lab report. So, can you help me out please? What is the issue here?

EspoTek, 2021-02-13:
Can you please give me the output of `lsusb` when the board is connected?

You can install it and run it through Homebrew:
```
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"
brew install lsusb
lsusb
```

hchaudh2, 2021-02-13:
I had already downloaded Homebrew earlier today. But this is what I got when I connected the board. 

>huriyyah@Huriyyahs-MacBook-Pro-2 ~ % lsusb
2021-02-13 00:12:00.434 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.434 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.435 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.435 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.436 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.436 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.436 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Bus 000 Device 001: ID 05ac:1018 Apple Inc. USB3.1 Hub  Serial: DLC01230154KD56A2
Bus 020 Device 001: ID 05ac:1017 Apple Inc. USB2.0 Hub  Serial: DLC01230154KD56A2
Bus 020 Device 003: ID 03eb:a000 Atmel Corporation USBGobindar Dummy Device 
Bus 020 Device 002: ID 05ac:1463 Apple Inc. USB-C Digital AV Multiport Adapter  Serial: DLC01230154KD56A2
Bus 128 Device 000: ID 05ac:8102 Apple Inc. Touch Bar Backlight  Serial: 0000000000000000
Bus 128 Device 000: ID 05ac:8302 Apple Inc. Touch Bar Display  Serial: 0000000000000000
Bus 128 Device 000: ID 05ac:027f Apple Inc. Apple Internal Keyboard / Trackpad  Serial: FM702860219N1R1A2+RWN
Bus 128 Device 000: ID 05ac:8103 Apple Inc. Headset  Serial: 000000000000
Bus 128 Device 000: ID 05ac:8262 Apple Inc. Ambient Light Sensor  Serial: 000000000000
Bus 128 Device 000: ID 05ac:8514 Apple Inc. FaceTime HD Camera (Built-in)  Serial: CC20272G4LHJ3Y328
Bus 128 Device 000: ID 05ac:8233 Apple Inc. Apple T2 Controller  Serial: 0000000000000000
Bus 000 Device 001: ID 1d6b:CITR Linux Foundation USB 3.1 Bus 
Bus 000 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub 
Bus 000 Device 000: ID 05ac:8102 Apple Inc. Apple T2 Bus 
huriyyah@Huriyyahs-MacBook-Pro-2 ~ %

hchaudh2, 2021-02-13:
I had already downloaded Homebrew earlier today. But this is what I got when I connected the board.

2021-02-13 00:12:00.434 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.434 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.435 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.435 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.436 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.436 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-02-13 00:12:00.436 system_profiler[10794:200635] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Bus 000 Device 001: ID 05ac:1018 Apple Inc. USB3.1 Hub Serial: DLC01230154KD56A2
Bus 020 Device 001: ID 05ac:1017 Apple Inc. USB2.0 Hub Serial: DLC01230154KD56A2
Bus 020 Device 003: ID 03eb:a000 Atmel Corporation USBGobindar Dummy Device
Bus 020 Device 002: ID 05ac:1463 Apple Inc. USB-C Digital AV Multiport Adapter Serial: DLC01230154KD56A2
Bus 128 Device 000: ID 05ac:8102 Apple Inc. Touch Bar Backlight Serial: 0000000000000000
Bus 128 Device 000: ID 05ac:8302 Apple Inc. Touch Bar Display Serial: 0000000000000000
Bus 128 Device 000: ID 05ac:027f Apple Inc. Apple Internal Keyboard / Trackpad Serial: FM702860219N1R1A2+RWN
Bus 128 Device 000: ID 05ac:8103 Apple Inc. Headset Serial: 000000000000
Bus 128 Device 000: ID 05ac:8262 Apple Inc. Ambient Light Sensor Serial: 000000000000
Bus 128 Device 000: ID 05ac:8514 Apple Inc. FaceTime HD Camera (Built-in) Serial: CC20272G4LHJ3Y328
Bus 128 Device 000: ID 05ac:8233 Apple Inc. Apple T2 Controller Serial: 0000000000000000
Bus 000 Device 001: ID 1d6b:CITR Linux Foundation USB 3.1 Bus
Bus 000 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 000 Device 000: ID 05ac:8102 Apple Inc. Apple T2 Bus

EspoTek, 2021-02-13:
I see the issue now: it enumerates as `Bus 020 Device 003: ID 03eb:a000 Atmel Corporation USBGobindar Dummy Device`.
There was a firmware issue with one particular batch of the boards produced last year, as well as a hotfix for this firmware issue in the newest version of the desktop software.

If you're able to connect your board to another PC running the latest version of the desktop software, the firmware will be fixed on your board and the board will connect in future.  If you don't have another PC, a VM should work too.

hchaudh2, 2021-02-15:
It worked! Thank you for that! Also what should I do if my multimeter in labrador is not working. I'm unable to click on anything like for example: I'm trying to change the mode to I, but I cannot do that.

hchaudh2, 2021-02-15:
It worked! Thank you for that!

EspoTek, 2021-02-15:
Good to hear, and sorry about the runaround.
Regarding the multimeter issue, read "A Note on Buffers" here:
https://github.com/EspoTek/Labrador/wiki

You'll need to turn off the oscilloscope before you can turn on the multimeter.

closed 2021-02-15 by EspoTek