espotek-org/Labrador #146: Issue with Multimeter

hchaudh2, 2021-02-15:
Also, another issue is coming up. When I click multimeter, everything goes away on the black screen and it sort of freezes up. I was trying to read the voltage and current after I made a circuit. What is the issue here?

EspoTek, 2021-02-15:
The Multimeter mode is a software wrapper around a special mode of the oscilloscope.  Scroll out on the display until you can see the trace, and the display should unfreeze.

hchaudh2, 2021-02-16:
I'm sorry I should have explained this better, but it doesn't necessarily freeze. The green and yellow on the black display disappears and then I can't measure the voltage or current anymore.

EspoTek, 2021-02-16:
Yeah, I understand what you mean.  You need to zoom out the display.

hchaudh2, 2021-02-16:
Sorry, how do I do that? I thought it was already zoomed out.

EspoTek, 2021-02-16:
Put the mouse cursor over the display and scroll out.  You should see a trace somewhere!

hchaudh2, 2021-02-16:
I think it worked for voltage, but when I try to get current. It says F for all values and the green/yellow disappears on the black screen.

EspoTek, 2021-02-16:
You're using 1.1, right?  Can I ask what version of MacOS you're running?

hchaudh2, 2021-02-16:
What do you mean by 1.1? My MacOS is version 11.2.1

EspoTek, 2021-02-16:
I'm talking about release 1.1 of the software.  It's about 3 years old now and has some pretty major bugs still in it.  Honestly, I can't offer support for such an old build.

If you're able to update your MacOS version to 10.12 or later, I strongly recommend doing that and upgrading to the continuous release of Labrador.
Otherwise, I'd recommend either installing Windows through Bootcamp or making a bootable USB of Ubuntu.

closed 2021-02-16 by EspoTek

hchaudh2, 2021-02-16:
I have updated my MacOS version to 10.12 or later, but when I update to the continuous release of Labrador, it won't open and ends up crashing.

EspoTek, 2021-02-16:
You can't run unsigned apps in 10.12 or later through a double-click.  Instead you need to right-click, select open, then confirm that you want to open it.

hchaudh2, 2021-02-16:
It still won't open when I do that.

EspoTek, 2021-02-16:
OK.  Can you please "Show Package Contents" on it, navigate to Contents->MacOS and then drag the raw executable in there onto a terminal window?  What does it say when you do that?

hchaudh2, 2021-02-16:
% /Volumes/bin\:Labrador/Labrador.app/Contents/MacOS/Labrador
QObject::connect: No such slot bufferControl::signalGenIn(bool)
QObject::connect:  (sender name:   'signalGenGroup_CH2')
QObject::connect:  (receiver name: 'bufferDisplay')
QObject::connect: No such slot isoDriver::setWindow(int)
QObject::connect:  (sender name:   'timeBaseSlider')
QObject::connect:  (receiver name: 'controller_iso')
QObject::connect: No such slot timedTickBox::resetTimer(bool)
QObject::connect:  (sender name:   'lockPsuCheckBox')
QObject::connect:  (receiver name: 'lockPsuCheckBox')
QMetaObject::connectSlotsByName: No matching signal for on_actionRecord_triggered(bool)
QMetaObject::connectSlotsByName: No matching signal for on_actionTake_Snapshot_triggered()
Attempting to open "/Volumes/bin:Labrador/Labrador.app/Contents/MacOS/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening "/Volumes/bin:Labrador/Labrador.app/Contents/MacOS/waveforms/Sin.tlw"
Length =  128
Divisibility =  4
List loaded!!
Attempting to open "/Volumes/bin:Labrador/Labrador.app/Contents/MacOS/waveforms/_list.wfl"
nameBuffer =  Sin

nameBuffer =  Square

nameBuffer =  Triangle

nameBuffer =  Sawtooth

nameBuffer =  DC
newName =  "Sin"
opening "/Volumes/bin:Labrador/Labrador.app/Contents/MacOS/waveforms/Sin.tlw"
Length =  128
Divisibility =  4
List loaded!!
Making USB Driver invisible!!
pipeID 0 = 129
Generic Usb Driver setup complete
unixUsbDriver created!
driver =  unixUsbDriver(0x7fc27669d000)
axes =  QCustomPlot(0x7fc276110c20, name="scopeAxes")

EspoTek, 2021-02-16:
Are you running 10.16 (Big Sur)?

hchaudh2, 2021-02-16:
It is Big Sur version 11.2.1

EspoTek, 2021-02-16:
Sorry, I'm an idiot!  I read it as 10.11 (El Cap).
Just enter the line `export QT_MAC_WANTS_LAYER=1` before launching from terminal and it should work perfectly.

hchaudh2, 2021-02-16:
Oh my god! It works! Thank you so much. I honestly really appreciate it a lot.

EspoTek, 2021-02-16:
No worries.  Hope you enjoy!

hchaudh2, 2021-02-16:
Sorry to bother you again, but when I click current on Multimeter, it says F for all values. Can you help me with this?

EspoTek, 2021-02-16:
What is your series resistance set to?

hchaudh2, 2021-02-16:
0 ohms

hchaudh2, 2021-02-16:
It works when I change it to for example 47 ohms but I'm not getting the proper readings for voltage, current or resistance.

EspoTek, 2021-02-17:
Check the docs.  They specify how to lay out a circuit for the multimeter, and what resistance ranges are sensible.
https://github.com/EspoTek/Labrador/wiki

Alternatively, get yourself an El Cheapo multimeter.  I still use my $10 Jaycar meter I got as a kid - they're simpler to use and very handy.