espotek-org/Labrador #148: Scope reads 4.5Vdc but multimeter reads 3Vdc

macktion, 2021-02-21:
Hi. I'm monitoring a small 4.5Vdc 1A PSU used on some tiny Christmas lights.  The Labrador zoomed-in shows quite smooth 4.5 (0v offset)but the multimeter shows only approx 3.15V max/min/RMS.  shouldn't the scope and meter values be close?  

I also haven't had luck measuring current with the Lab. trying to verify/improve the accuracy of the 200Ma I measured with my discrete meter that created a significant/unwanted series load that dims the lights. I could not even get the Lab to allow the string of lights to turn on while indicating current as high as 1.5A at 0ohms series.

EspoTek, 2021-02-21:
The multimeter measures the difference between CH1 and CH2.  If nothing is attached to CH2, it'll float towards VCC/2 (around 1.65V).
Please see documentation here:
https://github.com/EspoTek/Labrador/wiki

Current measurements with a low burden voltage is not a standard feature of most multimeters, including Flukes.
It's possible to use an external amplifier circuit to lower burden voltage to near-zero, though.
Dave Jones did a video on it a while back:
https://www.youtube.com/watch?v=jx0dryLmUEQ

closed 2021-02-21 by EspoTek

macktion, 2021-02-22:
Thank you.  Very helpful.

lkoepsel, 2021-03-27:
This issue is happening for me. If one follows the following process, it is easy to replicate.
1. Open Labrador application, press M to bring up Range Dialog and change Vmax to 5V.
2. Connect Oscilloscope CH1 to Power Supply (Positive) and Oscilloscope CH2 to Power Supply (GND) *Note the CH1 line is about 4.8V and is solid, no noise.*

[image]

3. Uncheck Oscilloscope CH1 and check Multimeter, all of the values (Max, Min, Mean and RMS) will be identical and incorrect. In this case, 2.96V. *Note CH1 line has now moved to 2.96V, continues to be solid, no noise*

[image]

4. Uncheck Multimeter and check Oscilloscope then check Paused and uncheck Paused. *Note the CH1 line moves up to 4.9V and once Paused is unchecked, the line appears with a slight amount of noise*
5. Uncheck Oscilloscope and check Multimeter and the values:
     * Are correct when compared with a DMM
     * Are not all identical, they vary with a little bit of noise.
 
[image]