espotek-org/Labrador #150: Driver is unavaiable

cthorsby, 2021-02-24:
I have tried restarting and updating my PC as well as using different cables. However, I keep getting the error message of the driver is unavailable.

EspoTek, 2021-02-25:
Hi mate, can you please send a couple of screenshots showing what you mean?  I'm not familiar with that particular error message.

cthorsby, 2021-03-01:
[image]
[image]

[image]

closed 2021-03-01 by cthorsby

EspoTek, 2021-03-02:
Did you solve the problem?