espotek-org/Labrador #152: Software Appears To Lock On Insufficient Bandwidth

mfoulks3200, 2021-03-04:
Just received my device in the mail (so excited), but it appears my device doesn't work with the hub on my bench (this is unsurprising, it's connected to a frankly insane amount of devices). The device appears to work fine when plugged in directly. My issue is, while troubleshooting this, it appears as though the program completely locks up when there is insufficient bandwidth. The program cannot be killed, even through task manager. Is there any way around this issue? Failing to a "device not connected" state after a certain timeout?

EspoTek, 2021-03-17:
Which OS are you using?

mfoulks3200, 2021-03-31:
Windows 10 Pro 64 bit Build 19041.804

EspoTek, 2021-06-26:
I did manage to get the waveform to appear glitchy by plugging the board in through an ungodly amount of hubs, but unfortunately can't reproduce the crash.
Since it doesn't seem to be affecting many other users, and since tracking down the bug is going to be near-impossible without access to your exact setup, I'm marking this issue as closed.

Apologies for that, and thanks for reporting the bug.

closed 2021-06-26 by EspoTek