espotek-org/Labrador #154: Pull data from multiple boards to have more channels to work with... Possible?

MrMaint, 2021-03-06:
I was looking at the Raspberry Pi shield that has 10 channels, but not as many other impressive features as the Labrador,  and wondered if the software could pull data from multiple Labrador boards to gain the use of 4, 6, or 8 channels. Would it be possible for the software to automatically recognize additional boards and show and coordinate more channels?

EspoTek, 2021-03-08:
This is definitely technically possible, but it's quite a niche use case and would be a significant undertaking to get up and running.  For those reasons, it's not something that I plan on implementing.

closed 2021-03-08 by EspoTek

techdragon, 2021-12-07:
@EspoTek does this mean the software can't work with more than one Labrador board?

EspoTek, 2021-12-08:
It means that it's technically possible, but you'd need to rewrite the software to do it.
You can only run a single instance of the official software, and you can only connect it to a single board.

techdragon, 2021-12-09:
@EspoTek Thanks for the reply! Not a lot of maintainers ever seem to notice comments on closed issues.