espotek-org/Labrador #156: Feasibility of dark mode?

matdombrock, 2021-03-12:
I know this seems trivial to some, but dark mode is an often overlooked accessibility feature. 

Plus look how cool it would be:

[image]

Is this something reasonably feasible?

EspoTek, 2021-03-12:
Genuinely curious, how does adding a dark mode make the software more accessible?

matdombrock, 2021-03-13:
I don't think a lack of dark mode will outright prevent anyone from using the software but it can make it very difficult and uncomfortable. 

For what it's worth I'm not aware of this being an "official" or protected disability. There is some debate on the validity of saying dark mode is an "accessibility feature" but I can say first hand it is. 

Some people just have very bad problems with eye strain, eye damage and blurry vision. For many people looking at dark text on a light background makes these problems a lot more pronounced. A lot of the time this can lead to bad headaches if they try to "push through it". 

For some people, blue light filters can help with the headaches but that doesn't help much with the readability of the text. 

Not exactly a scholarly source, but here is a good article from CGPGrey on the topic:

https://www.cgpgrey.com/blog/dark-mode-as-accessibility-feature

That all being said, I think I almost answered my own question about feasibility, because the way I got this app to be in dark mode is by changing the actual windows UI elements to be dark mode with an unsigned windows theme. 

I think without a total UI overhaul you would just need Windows to get it together with an actual dark mode UI at the OS level right?

EspoTek, 2021-03-17:
Oh, thanks for letting me know.  I had no idea that's why dark mode was such a big deal.  I'll leave this ticket open for now and get onto it ASAP.

EspoTek, 2021-06-04:
Just added it in, under the "Accessibility" menu.  Didn't realise how much more comfortable dark mode is.  Now going to change all my apps to do this.

closed 2021-06-04 by EspoTek

matdombrock, 2021-06-11:
Amazing. Thanks for all of your work on this project!