espotek-org/Labrador #157: help measuring current in multimeter mode

khtodd1, 2021-03-13:
Hello.  I'm trying to use Labrador to measure the gate current when switching on a MOSFET from an output of a microcontroller.  I've read the Wiki 500 times but I'm not still not quite sure how to connect it to measure what I want.  Can you please explain, or even better, throw up a quick circuit sketch?  Thank you.

EspoTek, 2021-03-17:
Unfortunately, I don't think it's a simple task for a device such as this to measure MOSFET gate currents as they're likely to be in the nA range.
However, if you could, the circuit diagram would like like this:
[image]
A load resistor is placed in series with the current you want to measure, and the two channels of the scope are added across the load.

The software interface looks like this:
[image]
Set the multimeter mode to "I" and manually enter the value of the series resistance (load resistor from circuit diagram) into the box shown.

Sorry if I can't be of more help.  If you're really keen to measure something that small, maybe consider looking to the uCurrent or similar?

closed 2021-03-17 by EspoTek

khtodd1, 2021-03-17:
Thank you for your help with this - it's a power MOSFET - so the gate
current could actually get up to 100mA or so, although I'd like to see it
on the order of 10-15mA.  I will try your suggestion.

EspoTek, 2021-03-18:
Ahh, of course.  You should have no issue then.