espotek-org/Labrador #160: Labrador Espotek not working

hchaudh2, 2021-03-16:
Hello, 

My labrador espotek is not working for my labs. It automatically says device not connected in the middle of the lab. My TA is saying to get a new one, but I'm not spending $80-$100 on a new one. Can you give me a refund so I can buy a new one that actually works?

Thank you

EspoTek, 2021-03-17:
If it disconnects mid-lab, it's not the board.  You're either shorting something out (most likely cause if the disconnects happen at the same point every time) or the cable is bad (most likely cause if the disconnects are random).

hchaudh2, 2021-03-17:
I used multiple cords and it had the same problem so the cable can't be the issue.  The cables I used also worked on other labradors as well (like my TAs). I don't think I'm shorting something because my TA made the same circuit on my labrador and on his labrador and his labrador worked and mine didn't. Any ideas?

EspoTek, 2021-03-17:
Oh, OK.  That's a very different story!

Please send an email to admin@espotek.com with your name and address and I'll get a free board shipped out to you ASAP, express postage.

closed 2021-03-17 by EspoTek

hchaudh2, 2021-03-17:
email sent!