espotek-org/Labrador #164: Labrador "How-To" for electronics neophytes

erandalln-dot-com, 2021-03-23:
For those of us who are new to electronics, a tutorial--video and/or written--on how to use the Labrador for each of its available functions would be great.  It would be especially good for newbs like myself are using your device as a economically accessible (and very "crowded bench" friendly) entry point in the subject.

lkoepsel, 2021-03-24:
I've been working on content that would do that. I literally just started last week, so I don't have much to show. I will have a series of lessons which takes a beginning user from installation to application using a simple inverting amplifier as a test case.

I hope I have the 9 steps covered by the end of the week and you can see if it begins to cover your needs.

www.wellys.com

erandalln-dot-com, 2021-03-25:
That's great.  I hope you can include something on timing pulses.  I'm doing Ben Eater's breadboard computer and realized I am completely clueless on how to determine what the duration of a momentary switch's pulse is.  It seems that would be really fundamental oscilloscope knowledge in digital circuit work.

I like your web site.  Looking forward to the progress.

EspoTek, 2021-05-25:
I have added a link to Lief's tutorial series to the main Readme.md.
It's just too good to let slide.

closed 2021-05-25 by EspoTek