espotek-org/Labrador #165: Duplication of Range Dialog

lkoepsel, 2021-03-27:
I like the idea of being able to set the Range Dialog default to automatically show. Unfortunately, when I do so on a Mac running Big Sur I get a two sets showing up. (See image) If necessary I can also test on Linux and Windows.

[image]

If it's possible while fixing/looking into this issue, it would be great if you could also have the values remain defaulted. Currently they also show up approximately as Vmax = 2.5V, Vin = -500mV. Retaining the new values in between application launches would be great.

EspoTek, 2021-03-29:
I have no idea why this appeared - I couldn't produce this behaviour on Windows.

Either way, I've just simplified the code that handles this dramatically so that only one `scopeRangeEnterDialog` object can exist at once.  It should fix the issue you were having - please let me know as I don't currently have access to a Mac for testing.

lkoepsel, 2021-03-29:
Nice work! Its been fixed on the latest release.

It would be nice if it could save the values as well.
Thanks!
Lief

closed 2021-03-31 by EspoTek