espotek-org/Labrador #168: labrador software not working on mac

hchaudh2, 2021-03-30:
I've been trying export QT_MAC_WANTS_LAYER=1  to open the labrador software, but it isn't working. It was working fine last week, but now it says: “Labrador” cannot be opened because the developer cannot be verified. macOS cannot verify that this app is free from malware.

Do you have any solutions for this?

EspoTek, 2021-03-31:
It's unsigned.  You'll need to right-click then open rather than double clicking.  On Catalina and later, you may also need to drag it into /Applications as well (I'm, still not 100% sure about this one).

lkoepsel, 2021-04-03:
I've been using the Labrador on macOS for quite some time. It works on both Catalina and Big Sur. As of the release in early March (approx), no additional commands are required to use the Labrador application on Big Sur.

A little more detail as to Chris' comments:
1. Download the latest release .dmg file.
2. Double-click it to mount it in Finder then drag to the Applications folder
[image]
3. Right click on the file and select Open.
[image]
4. The first dialog screen doesn't show the option to Open the file. Close the first screen, then do the same thing again. This dialog screen will allow you to open it.
[image]
5. You know you are successful when you see this screen:
[image]

I've got additional notes on the Labrador on my site: www.wellys.com

closed 2021-05-25 by EspoTek