espotek-org/Labrador #170: USB bandwidth - hardware or Linux device driver issue?

superfrink, 2021-04-10:
I have an older PC running Linux and am starting to suspect the USB hardware or the device driver for the USB controller does not support the bandwidth needed.

The Labrador device is detected but the isochronous initialization fails.

The Labrador application log contains:
```
newDig
libusb: error [submit_iso_transfer] submiturb failed error -1 errno=28
libusb_submit_transfer FAILED
ERROR LIBUSB_ERROR_IO
```

/var/log/kern.log contains:
```
usb 2-1.6: usbfs: usb_submit_urb returned -28
```

Kernel code and docs note 28 is ENOSPC meaning there isn't enough bandwidth to allocate what was requested.

The USB bus also has a mouse and built-in SD card reader.  There is a second bus with the built-in webcam.
```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 6: Dev 5, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
        |__ Port 7: Dev 4, If 0, Class=Mass Storage, Driver=ums-realtek, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 1: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 1: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
        |__ Port 2: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

I tried different USB ports.

It is an all-in-one design (computer/monitor/webcam/speakers in one case) so it is not the easiest to physically unplug the SD card and webcam.

I tried removing kernel modules (ums_realtek, usb_storage, uas) and echoing into ```sys/bus/usb/drivers/usb/unbind``` to remove the device.  That did not work.

I went into BIOS and disabled the SD card reader.  That didn't work.

I removed the mouse.  Still didn't work.  It looks like nothing else is on the hub or bus.

```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 6: Dev 13, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
...
```

Maybe the hardware or kernel driver doesn't support the bandwidth needed.  The SD card was listed at 480M on the same bus though.

Any other suggestions?

EspoTek, 2021-04-12:
Hi mate.  Looks like all the stars have aligned in terms of blind spots with the Atmel USB Stack (isochronouse endpoint size must be either a power of 2 > 8 or 1023 - I chose 1023 for Linux), the Linux USB stack (no mechanism for getting frame numbers from packets received) and your machine (uvcvideo on Bus 1, and the usbhid on Bus 2).

The fix, oddly enough, is to connect Labrador to your machine via a USB hub (without any other devices on the hub).  The hub communicates downstream to the PC at 480Mb/s while talking upstream to the Labrador at 12Mbit/s.  What hub you use shouldn't matter - even the El Cheapo AliExpress hubs now support 480MBit/s.

superfrink, 2021-04-15:
I tried using a hub built into a monitor.  No luck.

```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 6: Dev 4, If 0, Class=Hub, Driver=hub/2p, 480M
            |__ Port 2: Dev 5, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
```

I've had the monitor for over 5 years.  Maybe I need to try a newer hub.

Going in another direction, I read something about that may or may not be related about the code in Linux.  From https://linux-usb.vger.kernel.narkive.com/E2i3mQ9I/full-speed-1-1-device-fails-with-enospc-when-connected-to-non-root-2-0-hub

```
[It] generally behaves better than
the old scheduler. However the fact remains that both schedulers are
buggy. Neither of them is compliant with the USB-2.0 spec when
handling 1023-byte iso-IN transfers.
```

I don't know if that means 1024 might work any better than 1023 or not.  I haven't tried it.

EspoTek, 2021-04-15:
How old is the software on this PC?  I've not run into that limitation on newer versions of Ubuntu or Arch.

superfrink, 2021-04-15:
```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"

$ uname -a
Linux oldhp 5.4.0-56-generic #62-Ubuntu SMP Mon Nov 23 19:20:19 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

I tried it with and without the mouse.

Since I had been working on Android checked the firmware version but the application log shows it matches what is expected.

```
BOARD IS RUNNING FIRMWARE VERSION 0x0006
EXPECTING FIRMWARE VERSION 0x0006
FIRMWARE VARIANT = 0x02
EXPECTED VARIANT = 0x02
```

There are options in the BIOS to turn off individual USB ports.  Maybe turning the right combination off will help.

superfrink, 2021-04-15:
No luck after turning off ports for the webcam and card reader.

```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 6: Dev 3, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

I'm guessing it has something to do with the USB hardware and drivers.
```
Bus 002 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
```

Comments online indicate possibly some scheduler settings might help but it's not certain.

```
$ grep USB /boot/config-5.4.0-56-generic | grep CONFIG_USB_EHCI_TT_NEWSCHED
CONFIG_USB_EHCI_TT_NEWSCHED=y

$ grep USB /boot/config-5.4.0-56-generic | grep CONFIG_USB_EHCI_ROOT_HUB
CONFIG_USB_EHCI_ROOT_HUB_TT=y
```

Looking for that USB hub online I found https://bugzilla.redhat.com/show_bug.cgi?id=700681 that reads:

> WRT to the lsusb output this stands out:
> 
> Bus 002 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               2.00
>   bDeviceClass            9 Hub
>   bDeviceSubClass         0 Unused
>   bDeviceProtocol         1 Single TT
>   bMaxPacketSize0        64
>   idVendor           0x8087 Intel Corp.
>   idProduct          0x0020 Integrated Rate Matching Hub
> 
> So it seems that in its latest chipsets Intel has replaced the usb 1 companion controller with a usb 2 hub to provide usb 1 compat through the TT-s of the hub, and more shockingly it has implemented a single TT hub, so all usb 1 devices end up sharing bandwidth (as they woul have done with a companion controller).
> 
> This is likely the cause of your problems, unfortunately Linux support for usb 1 devices connected through a usb 2 hub is less then stellar. This is a well known problem with the ehci kernel driver. See the linux-usb archives for various discussions of this (search for me as author):
> http://vger.kernel.org/vger-lists.html#linux-usb
> 
> The -28 error the pwc driver is getting is the ehci driver saying it cannot schedule the iso stream requested on the TT in question, there is nothing the pwc driver can do here to fix this.

superfrink, 2021-04-17:
No luck with a different external hub on the PC.

```
$ lsusb -tv
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
        |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
            ID 050d:0414 Belkin Components 
            |__ Port 4: Dev 4, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
                ID 03eb:ba94 Atmel Corp. 
...
```

superfrink, 2021-04-17:
How come it works on a Raspberry Pi 3 B+ at 12M?  On the Pi the oscilloscope shows data.  On the PC after the isochronous mode fails to load there is no data on the oscilloscope.

```
$ cat /etc/debian_version 
10.9

$ uname -a
Linux p 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux

$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 2: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=lan78xx, 480M
        |__ Port 2: Dev 7, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
        |__ Port 3: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M

$ usb-devices
...
T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  7 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=03eb ProdID=ba94 Rev=02.00
S:  Manufacturer=EspoTek
S:  Product=Labrador
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=usbfs
...
```

EspoTek, 2021-04-17:
I honestly couldn't explain that.  I think it might be a quirk with your PC - I've not heard about the issue you're having from any other users, nor can I reproduce it on any of my machines.

superfrink, 2021-04-17:
Agreed, it does seem to be something with this one PC.

I tried a newer kernel (same version as the Pi).  No luck.
```
$ uname -a
Linux oldhp 5.10.17-051017-generic #202102170631 SMP Wed Feb 17 11:37:41 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

Grasping at straws, I compiled the Linux application using the same values as the Pi build for `ISO_PACKETS_PER_CTX` and `NUM_FUTURE_CTX`.  No luck.

EspoTek, 2021-04-17:
You're not going to have any lucking changing those variables without also updating the board's firmware - and then you'll be limited by Atmel's USB hardware (either 512 - too low, or 1023 - too high).
To be honest with you, it looks like a bug or driver limitation in the kernel that may not get sorted ever.  Can you get your hands on another old PC or is this the only one?

closed 2021-05-25 by EspoTek