espotek-org/Labrador #171: Can't connect on fedora linux or android... works fine on macosx

alvaroc1, 2021-04-13:
Getting this when I run the software on linux (Fedora 32):
```
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
Libusb context initialised
DEVICE NOT FOUND
Connected
```

:heavy_check_mark: On macosx: `lsusb` lists it just fine the software picks up and works great. 
:negative_squared_cross_mark: On linux: Does not get listed when I use `lsusb` on linux and the software shows "Device not :negative_squared_cross_mark: connected!"
On android: Doesn't get picked up by the app when connected to my huawai mate se phone (android using USB-OTG).

What could be the problem?

alvaroc1, 2021-04-13:
Never mind. It turned out to be a bad usb cable. Sorry for the noise.

closed 2021-04-13 by alvaroc1

EspoTek, 2021-04-13:
All good.  Thanks for closing the thread when you solved the issue.  :)