espotek-org/Labrador #175: Licensing issues

mikhirev, 2021-05-06:
Hi!

Your project does not contain copyright information and license text. README.md states that all software files are licensed under GNU GPL v3, but this is wrong at least for firmware that contains source code copyrighted by Atmel and licensed under the terms of non-free license.

Please provide more detailed and correct copyright/licensing information.

EspoTek, 2021-05-07:
I am not a lawyer and have no idea how to resolve this.  If you would like to submit a PR rectifying this, go ahead and do so.

closed 2021-05-07 by EspoTek

mikhirev, 2021-05-07:
Well, I can add the GPLv3 text and copyright notices (are you the only copyright holder?) to all source files as [recommended in the GPL itself](https://www.gnu.org/licenses/gpl-3.0.html#howto). But I cannot choose a proper license for firmware, only give some suggestions.

As I see, the Atmel license is based on the [3-clause BSD license][BSD-3-Clause] but contains an additional restrictive clause, making it non-free and incompatible with GPL. It means that you are not permitted to distribute the compiled and linked firmware under the terms of GPL. If you distribute only the source code, licensees could compile and use it, but also couldn't redistribute .hex files.

You may choose for your code the unmodified 3-clause BSD license or the [2-clause BSD license][BSD-2-Clause] if you don't care about using your name in promotions. Then you and every licensee will be able to (re)distribute .hex files under the terms of Atmel license because it is more restrictive.

You may also apply the Atmel license to your code, but I don't see any point in doing so. You may choose another permissive license for your code, but if you do, you will need a lawyer to conceive resulting terms of .hex files distribution.

[BSD-3-Clause]: https://opensource.org/licenses/BSD-3-Clause
[BSD-2-Clause]: https://opensource.org/licenses/BSD-2-Clause

mikhirev, 2021-05-08:
If you want to distribute your code under GPLv3, it is also possible if you add a license exception as described [here](https://www.gnu.org/licenses/gpl-faq.en.html#GPLIncompatibleLibs).

EspoTek, 2021-05-09:
OK, thanks for the explanation.  I've updated the readme to state that the Microcontroller side of the software is licenced under the 3 clause BSD license.