espotek-org/Labrador #179: What is Vcc for the Labrador?

mistere357, 2021-05-15:
I'm looking for Vcc info for the Labrador board ( Min - Nominal - Max) 
I've read through the documentation so if it's there I just missed it :frowning: 
A nominal 5.0 V DC makes sense however as a source it can output 12 V DC.  Boost converter?
Thank you

EspoTek, 2021-05-16:
I'm not quite sure what you mean by this question.  You should not be supplying any additional power to the board other than the USB input.

mistere357, 2021-05-16:
There are two pins that plug into the COM and + rows of the breadboard. 
If I don't have to do anything with them that's great. 
Thank you

closed 2021-05-16 by mistere357

EspoTek, 2021-05-17:
They're output pins, not inputs!  No additional hardware required.

mistere357, 2021-05-17:
Yes, as soon as you said all I needed was the USB I understood! :bulb: 
Great product by the way - I'm having fun tinkering with it. 
I bought 2 so I'm going to set up the 2nd one on my laptop (1st is on the Raspberry Pi) and use the outputs from one as input to the other :grin: 
Thank you again!

EspoTek, 2021-05-18:
Awesome, hope you enjoy!