espotek-org/Labrador #180: How to use voltage > 3.3V?

gigamegawatts, 2021-05-19:
Could you please explain what connections are needed to work with a voltage of, say 5V, for the Waveform Generator and Oscilloscope?

Right now, output from the Waveform Generator is clipped at 3.3V.  I'm afraid to connect a 5V signal to the oscilloscope without knowing what will happen.

I'm guessing that it has something to do with moving the "Power Supply On/Off Header" jumper and supplying voltage through some pins (Power Supply Pins or maybe the Expansion Header), but I don't want to guess wrong and fry something.

Thanks,

Dan.

EspoTek, 2021-05-19:
Don't touch the header.  Just increase the power supply voltage through the software interface to at least VPeak + 2V and the signal generator won't clip.

Nothing needs to be done to get up to 20V on the oscilloscope.  The frontend handles all of that and prevents blue smoke release.

closed 2021-05-19 by EspoTek

gigamegawatts, 2021-05-19:
Yes, that works, exactly as you described.  

By the way, my Labrador is one of the old "Batch 1" units, bought in 2017.  It's great to see that it's still fully supported and works with the latest firmware.  Thanks for that!

EspoTek, 2021-05-20:
Of course.  It doesn't seem right to stiff the original buyers - you guys put the money up so that EspoTek could launch.

Glad it works and hope you enjoy!