espotek-org/Labrador #181: Unable to resize app on 7inch display

zabolca, 2021-08-02:
I apologize if the fix is out there and I just cant find it. I just installed on a raspberry pi 4 with a 7 inch, touch, LED display and I'm unable to resize the app or reach the bottom of it. I have tried hiding the widgets in the menus but the graph just ends up taking up the room. I was unable to find a config file to adjust as well.

EspoTek, 2021-08-02:
What's the resolution of the display?

zabolca, 2021-08-02:
I currently have to jump it up to 19201080 to view the entire app but that
makes the rest of the desktop hard to see.
I think originally it started at 1024x600 but 1280x960 isn't too bad but I
still can't see everything.
Is there a way to resize anything in it? If I could take out what i
dont need at the moment, would that be great?

EspoTek, 2021-08-02:
You'd need to edit MainWindow.ui and recompile to get it running at 1024x600.  The minimum supported resolution is 1366x768.

zabolca, 2021-08-03:
Ah ok, I've never compiled anything, I'll look into it.
Thanks

closed 2021-08-04 by EspoTek