espotek-org/Labrador #190: Multimeter readings in V mode

fijam, 2021-10-31:
I have an issue with the multimeter in V mode.

With nothing connected, the multimeter stats show approx max 200mV, min 100mV, mean 145mV.

[image]

I have performed oscilloscope calibration and it looks fine, when I connect oscilloscope DC channel 1 and 2 to the USB shield I get 4e-05 mean on both. Disconnected both channels float at ~1.65V. But as soon as I switch to multimeter mode the mean is ~145mV. 

So for example, if I connect the mutlimeter CH1 to PWR and CH2 to GND I get a reading of 4.95V. If I connect CH2 to PWR and CH1 to GND I get -4.7V instead.

EspoTek, 2021-11-04:
That's a really interesting and quite concerning quirk!  It's not something I've seen or heard of, but will look into it when I have time (probably around Christmas)

EspoTek, 2022-01-24:
I've tried to reproduce this and am unfortunately unable to.
I'm wondering if this was one of the quirks introduced by the performance-related bug (which only showed on low-end machines and is now fixed)?

closed 2022-01-24 by EspoTek