espotek-org/Labrador #191: Labrador Turning Off

lukegolding123, 2021-11-05:
Hi,

When I open the labrador software and plug in the labrador, the red light turns on for about 2 seconds and then switches off. The only thing plugged in is the micro-usb cable. The software also says the device is not connected. 

The board was working fine for about 2 months, and this has happened suddenly. Any suggestions?

EspoTek, 2021-11-07:
Is there any circuit attached to the Labrador?  It sounds like the fuse is tripping.

lukegolding123, 2021-11-08:
There’s nothing attached, yep looks like the fuse is tripping to me too. Although not sure how as there is nothing attached to the board, and it doesn’t look like it’s been damaged.

EspoTek, 2021-11-08:
OK, try putting a blob of solder over the fuse bypass header:
https://github.com/EspoTek/Labrador/wiki/Pinout

If it explodes after you do this, contact me via email (admin@espotek.com) and I'll send you a replacement free of charge.

closed 2022-01-24 by EspoTek