espotek-org/Labrador #197: fail to install on the new raspberry pi OS Bullseye

Ethanali, 2021-11-15:
can't get qt5-default

EspoTek, 2021-11-15:
Thanks.  I'll look into this when I have the time.

A quick google suggests the following should function as a workaround:
`sudo apt-get install qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools`

~Chris

jsmittelman, 2022-01-17:
> Thanks. I'll look into this when I have the time.
> 
> A quick google suggests the following should function as a workaround: `sudo apt-get install qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools`
> 
> ~Chris

tried that, still cannot locate qt5-default

EspoTek, 2022-01-18:
You don't need qt5-default.  Just remove the line from the script.

EspoTek, 2022-01-24:
This is now fixed.  The updated script has been tested on both Stretch and Bullseye.

closed 2022-01-24 by EspoTek