espotek-org/Labrador #198: Add menu shortcuts

skorokithakis, 2021-11-29:
Can we get some shortcuts to the menu items? They're very discoverable and it would be very handy to be able to "snap to cursors" etc quickly.

EspoTek, 2021-12-27:
Hey man, great to see you're still using this, and thanks for the plug on HN.  :P

Did you have any thoughts as to what the options could be?
I was thinking maybe Ctrl-1/Ctrl-2 for Snapshot CH1/CH2, Shift-1/Shift-2 for DAQ CH1/CH2, and middle-click for Snap to Cursors?

Anything else?

skorokithakis, 2021-12-27:
Hahah well, I love it, and the plug is well-deserved (I plug it every chance I get :P).

I'll reply with ideas shortly when I'm at the PC so I can look at the menu!

skorokithakis, 2021-12-27:
Hey Chris! I'm back at the PC. Snapshot/DAQ sounds good, but I don't know if "snap to cursors" should be a middle click (maybe that should reset the view somehow?). In general, I've had a few issues with the zooming, so I wanted to open a different ticket.

Maybe Snap to Cursors/Enter Manually could be Ctrl+Shift+S/M?

mi-hol, 2025-02-10:
@skorokithakis is this still an open issue?

mi-hol, 2025-02-10:
linked with #111 ?