espotek-org/Labrador #200: Raspberry Pi 4b - Labrador - "Calibration has been abandoned due to out-of-range values"

fluxsauce, 2021-12-20:
Good news - I can calibrate with Windows 10 without issue! I'm assuming the unit itself is fine.

Difficulty - I can't get it to calibrate on a Raspberry Pi 4b, Raspberry Pi OS with desktop 5.10.

The device detects and connects. The software seems to be working correctly, other than this calibration step?

I had to struggle a bit to get it installed, https://github.com/EspoTek/Labrador/issues/197 mentions it and the workaround is what I did (honestly, after Googling first, then I found the issue 😆)

Raspberry Pi firmware is current, upgraded as part of this process. I also used a breadboard to ground it, same effect, so it's not an intermittent connection. I'm using a dedicated USB-C power supply (Canabalt) connected to a nicely grounded circuit so I'm assuming that's not it either.

Steps to reproduce:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Thanks!

EspoTek, 2021-12-20:
Any idea what those big bursts of noise are on the trace in pics 1 and 2?

fluxsauce, 2021-12-20:
Breadboard jumpers were connected to the Labrador.

Nothing connected except USB cable:

https://user-images.githubusercontent.com/976391/146708428-9f58b94f-8845-4714-9db5-1380510ebee9.mp4

EspoTek, 2021-12-27:
OK, that makes more sense.  Looking at the log, it seems like it detected a higher value than expected in the second stage.  Perhaps you lost contact temporarily, just before pressing OK?

closed 2022-01-03 by fluxsauce

lchen198, 2022-01-16:
I am having the same issue on raspberry pi 4 and tried multiple times without luck.  It works fine on Win 11.

...
genericUsbDriver dectructor entering
genericUsbDriver dectructor completed
Reinitialising USB driver!
Making USB Driver invisible!!
pipeID 0 = 129
Generic Usb Driver setup complete
unixUsbDriver created!
ReinitUsbStage2 is returning
deviceConnectedDisplay::connectedStatusChanged running!
CHECKING CONNECTION!
Entering unixUsbDriver::usbInit
Libusb context initialised
Device found!!
KERNEL DRIVER INACTIVE
Interface claimed!
Connected
BOARD IS RUNNING FIRMWARE VERSION 0x0006
EXPECTING FIRMWARE VERSION 0x0006
FIRMWARE VARIANT = 0x02
EXPECTED VARIANT = 0x02
Connecting now!
deviceConnectedDisplay::connectedStatusChanged running!

...

unified debug is of size 27
unified debug is of size 27
unified debug is of size 27
VMeasured (CH1) =  0.466232
VMeasured (CH2) =  0.467766
unified debug is of size 27
unified debug is of size 27
unified debug is of size 27

StriderTR, 2022-01-18:
I'm having the same issue, mine seems to read a constant 1.6V, but fails at the grounding. I tried grounding several ways, even to my PC's PSU.

I am also using mine with a Pi 4, same setup as the OP. My Pi 4 is in a Argon M.2 case. If I figure it out, I'll post it. :)

theblaz, 2022-01-18:
Having the same issues but with the Linux AppImage on an x86 computer. Could there be an issue with the Linux reading values? Calibration works from windows application

EspoTek, 2022-01-18:
I've had several reports of this bug across all platforms, and can't for the life of me reproduce it on any of my systems at home.  It seemed to appear about 2-3 months ago, and can disappear randomly too.  I'm not sure what the source of this could be, but it might be best to just roll back some patches from the past few months.

theblaz, 2022-01-18:
thanks for the update. I'll keep playing with it.

lchen198, 2022-01-18:
Maybe try a different USB cable?  I haven't tried it yet but I wonder if connecting the ground pin on the board with raspberry pi ground pin would help.

theblaz, 2022-01-18:
I tried several different cables and different ports as well as different
machines. The consistency seems to be Linux as I am able to get it to work
in windows. I haven't tried it on a Mac yet.

Sent from my mobile device, please excuse brevity, spelling and punctuation.

StriderTR, 2022-01-19:
Nice to see it's not just me. 

I tried 4 different USB cables as well as different grounding methods. I  have not tried on my Windows machine yet, haven't had time. 

I can say that the "noise" you see in the OP, I see as well, I just assumed it was when my skin touched the wires. That's what led me to try different grounding methods, I was assuming it was seeing static through those thin jumper wires. Still, same problem.

EspoTek, 2022-01-19:
The noise is a separate issue, and not something I've seen outside of the Pi (again, not critical and not something I'm able to reproduce with my Pi 3, so I haven't spent too much time looking into it)!

The common thing that people seem to be is the trace freezing after a few seconds, then coming back after the scope is enabled/disabled.

StriderTR, 2022-01-19:
Would it have anything to do with how the Pi is grounded? If I am not mistaken, the Pi grounds via HDMI to your monitor/TV. Not the same as your PC. Though I tried grounding mine to my PC and PSU chassis and got the same result. I'm just spitballing ideas, it's been a long day. lol

EspoTek, 2022-01-19:
Genuinely no idea.  "Hard to reproduce" and "not critical" means I'll probably never know, sorry!

theblaz, 2022-01-20:
I downloaded the Linux Legacy Appimage and I was able to calibrate no problem. Thanks for the suggestion and help.

StriderTR, 2022-01-20:
Sadly, I don't see a way to get a legacy app image for Pi. Is it available anywhere?

Also, I did ground my Pi to my PC case with one of the GPIO grounds, it did affect the reading, but calibration still failed. Oh well, it was an idea. 

[image]
[image]

EspoTek, 2022-01-20:
Finally had a bit of a lightbulb moment!
I think I know if not why, but at least when the issue was caused, and that it's part of the software.  I'll get a patch ready over the coming few days to roll back some of the changes that may have caused this regression!

StriderTR, 2022-01-20:
Excellent! I'll give it a try once it's available. :)

EspoTek, 2022-01-23:
It should be fixed now (keyword: should)!

Long story short, the patch that added the frequency spectrum was doing a whole lot of unnecessary maths when it wasn't enabled and overloading the Pi's tiny CPU, causing it to miss timing contraints.

I've pulled this out of the regular path (it now only runs when the spectrum mode is enabled) and it seems to be back to normal, at least from my end.

If other people are able to confirm this, I can fully close this off once and for all!

StriderTR, 2022-01-23:
Calibration Successful!

Thanks for the fix! The Pi's tiny processor, and I, thank you! :P 

Perfect timing, the rest of the parts I ordered are arriving today so I will be able to give the Labrador a proper home. BNC connectors and probes for the scope, Banana connectors and probes for the multimeter and logic, and a clamp speaker type terminal block for the signal generator. All going in a simple black project enclosure. Once I have it all done I can take a nice pic and drop a proper review on Amazon.

Again, thanks for all your help! It's been greatly appreciated.

[image]