espotek-org/Labrador #201: Feature request: Middle click drag?

skorokithakis, 2021-12-27:
I'm frequently having issues panning to the right range, because there's no way to drag the view to where I want it. For example, to show -5V to 5V, I will have to zoom out so the entire range is in view, position my mouse pointer strategically somewhere and then zoom in until the range is in view (or use the "enter manually" menu item).

Would it be possible to middle-click-drag the view to pan? I think I opened another ticket about this a while ago but I couldn't find it, so here I am again :P

MatthewSteeves, 2023-02-21:
I know this is old, but in case still a pain point: Have you explored "Snap to cursor" and it's keyboard shortcut?

skorokithakis, 2023-02-22:
@MatthewSteeves hmm, it's been a while so I'm not sure if I have, but I'll try that, thanks for the help!

skorokithakis, 2023-04-02:
@MatthewSteeves I just tried it, I didn't realize there was a keyboard shortcut, thanks! I still think that middle-click-drag would be very useful, but at least the shortcut takes some of the pain away.

mi-hol, 2025-02-10:
linked with #111 ?