espotek-org/Labrador #202: Potential improvements to spectrum view

tdunning, 2021-12-30:
The new spectrum view feature is nifty, but very limited.

Here is a list of suggestions:

* double sample rate apparently suppresses the spectrum analyzer. It would be nice if it could provide more frequency resolution
* I can't find a way to change the frequency scale. For the low frequency signals that the Labrador normally works with, this means that the display is nearly useless
* the trigger status display obscures some of the spectral display. Turning off the trigger is a work-around, but not at all obvious because the trigger seems to have no effect on the spectral display
* it would be good to do a sin^2 or sin^4 window on the signal to suppress artifacts (mostly useful if you can change the frequency scale) 
* the vertical scale is undefined. It appears to be a percentage of the highest value in the spectral display. A log scale calibrated against dbmV would be much more useful for actual measurements.
* the vertical scale can't be adjusted. This is important if you are looking at something other than the biggest peaks, especially if changing the horizontal scale becomes possible

Vincenzo-Petrolo, 2022-05-21:
Hello @tdunning , thanks for all those comments!
I wrote the spectrum feature one year ago, but had no time to fix all the GUI related things. 
I'm starring your issue in order to implement those requests in the future. As I am not a pro in User Interfaces, it would be really appreciated if someone could fix them!

EspoTek, 2022-05-29:
> I wrote the spectrum feature one year ago, but had no time to fix all the GUI related things.

And I never sent you that letter!   One sec, man!

mi-hol, 2025-02-10:
@tdunning @Vincenzo-Petrolo is this still an open issue?

tdunning, 2025-02-11:
I have no idea. I haven't used the Labrador in quite some time (I bought a $500 oscilloscope, a tinySA and a nanoVNA).

Vincenzo-Petrolo, 2025-02-12:
Hi there, yes it is still open (unfortunately :-( ). 

Currently, I have no bandwidth to deal with it so I think new contributions would be welcome!

mi-hol, 2025-02-13:
Thanks for your feedback.

It seems splitting up the initial comment into multiple sub-issues is required to separate bugs from new features.
Would perhaps someone with technical insight volunteer to perform that task?

mmehari, 2025-02-13:
Although I didn't fully understand the second point (_way to change the frequency scale_), I totally agree splitting points into new features and bug fixes.

For example, the 1st (_double sample rate_) and the 4th (_windowing function_) could be considered as new features where as the remaining ones could be considered as bug fixes (_trigger status_, _vertical scale unit/adjustment_).

The new features might need a fresh pair of eyes but the bug fixes could be best handled if first commented by @EspoTek

tdunning, 2025-02-13:
The "changing the frequency scale" item was about changing the start and end frequencies.

tdunning, 2025-02-13:
I will file additional issues and reference this as the uber issue.

tdunning, 2025-02-14:
See #286, #287, #288, and #289 

for the broken out bugs related to this issue.

mi-hol, 2025-02-14:
> See [#286](https://github.com/espotek-org/Labrador/issues/286), [#287](https://github.com/espotek-org/Labrador/issues/287), [#288](https://github.com/espotek-org/Labrador/issues/288), and [#289](https://github.com/espotek-org/Labrador/issues/289)
> 
> for the broken out bugs

Thanks for investing the time, greatly appreciated!

> I can't actually link the issues to this one due to low privileges on this repo.

Not sure to understand that, are you not having access to the button "create sub-issue" at bottom of first comment as I do?

[image]

mi-hol, 2025-02-14:
@tdunning I tried to link the new issues via "edit" ing the first comment text. 
Please verify and correct in case I got something wrong.

tdunning, 2025-02-14:
I don't see the "create sub-issue" button. The "relationship" stuff on the right is also greyed out for me. This is typical for a non-owner of a repo and not a big deal.

Your edits to the first comment look great. I checked that all the links go to the right issues.

mi-hol, 2025-02-18:
> The "changing the frequency scale" item was about changing the start and end frequencies.

@mmehari did this help to address your comment?

mmehari, 2025-02-18:
It could be that the bug was issued long time back and recent changes resolved it quietly but there are start and end frequencies in the frequency spectrum view. Didn't I get it correctly or it is not an issue any more?

tdunning, 2025-02-18:
In the breakdown, I combined the need to zoom in both frequency and amplitude into a single issue.

Both may or may not be resolved at this point. I don't have a current version to look at.