espotek-org/Labrador #204: Windows Installer - Virus Detection

StriderTR, 2022-01-11:
Just an FYI. 

Kaspersky Security Cloud (Ver. 21.3.10.391 - Database 1/11/22) reports a "Trojan.Win32.Redyms.vle" detection on the Windows Installer (Labrador_Installer.exe). 

Likely a false positive, but just in case, I wanted to make you aware. I just bought it and was in the process of hooking it up to test it out and got this hit.

Thanks!

EspoTek, 2022-01-11:
Yep, I'm aware of this.  From time to time, various Antivirus programs will detect Labrador as a false positive.  I'll then submit the sample, wait 24 hours and it's usually all good.

However, Kaspersky doesn't allow regular developers to submit samples larger than 50MB (the Windows installer for Labrador is around 70MB).

It's not just EspoTek that's affected by this!

For this reason, I'd recommend switching to a different Antivirus provider.  I've personally had no issues with the built-in Windows AV and a bit of common sense (ie not downloading random .exe files from spam email attachments and running them).  I suspect if you're technical enough to use an oscilloscope you'd be in the same boat.

closed 2022-01-11 by EspoTek

StriderTR, 2022-01-12:
I just told it to ignore it. I plan on running the software on my Pi 4 anyway, once my BNC connectors arrive and I can get this project done. 

Also, I was not aware of Kaspersky's limit, that does indeed suck.  =/

Anyway, thanks for the response and keep up the excellent work!

EspoTek, 2022-01-14:
Thanks mate.