espotek-org/Labrador #207: Power by switch - QUESTION

bpod, 2022-01-14:
Does anyone have advice on how I would wire the power up to a switch to easily turn the board on/off?

Thanks in advance.

EspoTek, 2022-01-15:
You could quite easily wire a switch into the +5V rail adjacent to the USB connector, but I'm honestly not quite sure why you'd want to.  Is there a reason you can't just unplug the board after use?

bpod, 2022-01-16:
Nope.. Was mainly just curious. Was thinking of building a case that is always plugged in. With leads, switches, etc.

EspoTek, 2022-01-16:
Ahh, gotcha!  That makes sense, thinking about it.  Let us know how you go if/when you get it working.  Just cut the +5V trace with a screwdriver or dremel, expose some copper on either side of the break and wire in the switch.  Classic bodge work!

closed 2022-01-16 by EspoTek