espotek-org/Labrador #209: Getting pins into the breadboard

johnchurchill, 2022-01-27:
Before I risk damaging my brand new Labrador, can someone advise if the attached picture is normal and I should be bending the PSU pins so far to get into the power rail? My board says "Batch 2" if that matters.

[image]

EspoTek, 2022-01-27:
Yep, it's fine.  The measurements are slightly off on some breadboards.

closed 2022-01-27 by EspoTek

jimustafa, 2022-04-23:
I have a "Batch 2" board as well; but, in my case, the power supply pins appear to be missing. Is there something wrong with the board?

[image]
[image]

EspoTek, 2022-04-23:
That is a manufacturing error.
Please contact me at admin@espotek.com and I'll organise a replacement for you.