espotek-org/Labrador #211: Noise - example, expectations

usrlocalben, 2022-02-02:
Here is a 15khz 50fps video Vsync/Hsync pair (trigger on Vsync).

I expect some temporal noise on the Hsync signal since it's getting into the range of the 325khz limit, but the measurements that are well below 0V and above 3.3V are surprising.  On my Rigol DS1052E they are very clean.

Do I have a bad copy?

[image]

usrlocalben, 2022-02-02:
Maybe a related or better question is, why does CH1 appear to affect CH2?
Below is CH1 on VSync, CH2 not connected to anything.

[image]

EspoTek, 2022-02-02:
Honestly mate, the answer to both of your questions is "it's a low-cost device that was designed for beginners to explore basic circuits".  When you really push the limits you will start to see cracks.

You're seeing the first lot of noise because the device uses the microcontroller's internal PGA as a ghetto low-pass filter, rather than adding dedicated hardware (and cost).  To be honest with you, I probably wouldn't make this decision again.  It really only knocked off 5c or so from the BOM, and HF noise does bite you in the arse when dealing with short impulses like this.

The second issue comes from the fact that both input share a single ADC (including same S&H cap) and analog frontend.  This will of course cause noticeable coupling when one channel is left to float, but it will be much less apparent if you drive CH2.

closed 2022-02-02 by EspoTek