espotek-org/Labrador #212: Raspberry Pi 4 (updated Buster) installation terminated unsuccessfully

wwoofbum, 2022-02-28:
After "Cloning into 'labrador'...
remote: Enumerating objects: 779, done.
.
.
Receiving objects 100% (779/779) ,  reused 662 (delta 197), pack-reused 0
Resolving deltas: 100% (209/209), done.
qmake could not find a Qt installation of ''
~$

[fullListing.txt](https://github.com/EspoTek/Labrador/files/8157337/fullListing.txt)

bobhb, 2022-03-13:
Same issue. Apparently systemic. Holding off on bad review at store. Chris, a bone please?   Thanks!!

EspoTek, 2022-03-14:
[image]

I'll try and reproduce it tomorrow.  This is on Buster, right?

EspoTek, 2022-03-14:
Have reproduced it now.  Got the exact error you did, and it seems like a one-line fix I made to the bootstrap script has made the issue go away!

EspoTek, 2022-03-14:
Yep.  Looks like that did it.  Software successfully compiles and runs.  Could you please let me know if it fixed it from you end, then I can mark this one off as done!

bobhb, 2022-03-14:
Good one. Glad you found it!

EspoTek, 2022-03-14:
Awesome.  Marking as closed for now.  Please let us know if more issues pop up!

closed 2022-03-14 by EspoTek