espotek-org/Labrador #213: 'Batch 1' board "bricked" - scope not running

DeeJay, 2022-03-06:
I have a Batch 1 board purchased March 2018. I seem to have 'broken' it.

Yesterday I had it working with the current (Jan 2022) Windows continuous release of the Labrador software. (Should  have been content with that... )

Then I tried a build of Labrador for RasPiOS Bullseye 64-bit running on an RPi4B2G board. The Desktop interface appeared but reported "Device not detected". I did not investigate this in detail. 

Since then I have been unable to use the Labrador board with the previously-working MS Windows installation. I have (apparently) successfully forced a Firmware Recovery by grounding the required pin and using the Help menu Recovery  option. But the 'scope function doesn't seem to run ('random' noise appears briefly then freezes) and attempts to calibrate always fail with a report of readings out of range. 

I have also tried reverting the Windows software to the legacy Update 1  and Update 2 releases,  but with no improvement. 

I think I have read all the relevant Documentation and tried all the Troubleshooting and Recovery tips. It feels as though  I have got the on-board processor into a state where it is no longer running correctly but cannot be reset?

Is there anything else I can do to breathe life into the little guy, or is this going to be a 29$ fix?

EspoTek, 2022-03-06:
Hi mate, sounds like a bog standard bad flash.  Completely recoverable.

If you can get the board into bootloader mode, check out the two batch files under C:/Program Files/Espotek/EspoTek Labrador/Firmware

Run wipe.bat, then flash.bat and it should spring back to life.
If not, remove the wires that force it into bootloader mode then reconnect it twice.

If that doesn't fix it, shoot us an email and we can work it out together.

closed 2022-03-08 by EspoTek

DeeJay, 2022-03-11:
Just a brief comment to acknowledge that I mis-diagnosed my own problem.

I can confirm that the board itself is fine: the inability to run the 'scope - or any other data acquisition - seems to be a quirk in running the Windows interface on my Windows10 laptop.

I am in touch with  Chris via email to try to get a better understanding of why my HW/OS platform isn't being cooperative.