espotek-org/Labrador #215: new board, after calibrate channel 1 always read around 1.667 volts and multimeter reads new 9v Alkaline at 10.1v

Mclute0, 2022-03-22:
am I missing something or am I getting stray voltage from somewhere?

my cheap multimeter reads the 9v battery at 8.7v, pretty sure 10.1v is just wrong.

AC coupling brings channel to zero.

the voltages below are all means.

touching the channel one probe to the USB ground show -0.12v

touch channel on to channel 2 no changes both on they both read 1.667v.

channel one to board + rail show 4.748v

channel one to - rail shows -0.03v

channel one multimeter to board - rail shows -1.6v with channel two to + rail also read 0.04 
channel one multimeter to board + rail shows 3.16v with channel two to - rail also reads 4.84

both channels read a ~60hz background oscillation at around .01 volts.

EspoTek, 2022-03-22:
Hi mate, can you please attach a screenshot showing the 10.1V battery?

Mclute0, 2022-03-22:
when first connecting the 9V battery, it always reads 3V for all three with no swings, but when I use the mouse wheel to scroll in or out it starts reading the 10V on a new battery and 7.9V on an older used battery with a +-0.02 volt swing.

[image]

EspoTek, 2022-03-23:
Ahh, I see.  Make sure the trace is visible on the display; otherwise you may get issues with the amplifiers clipping.

EspoTek, 2022-03-23:
And just to confirm, this is with CH1 (DC) on the positive terminal, and CH2 (DC) on the negative?

Mclute0, 2022-03-24:
yes,  CH1 (DC) on the positive terminal, and CH2 (DC) on the negative. I also have not been able to get the interface to show uF. I,V,R show up, but changing to C and I get no F listed in the output stays at whatever the last reading was liike mV mA or resistance.

maybe I will take a look at the code and see if I see any issues, I used to do a little debugging for a living.

Mclute0, 2022-03-24:
I is now giving F? without the μ? no numbers are sowing but minus sign comes are goes.

I see the display code and it looks pretty clean. I would guess this is a problem with the redraw data on window change procedures, that's without looking though all the code. still looking... The debug is showing cap_x1 == -1 that I take means no/bad reading from the board but it is transferring data of some kind.

[image]

EspoTek, 2022-03-24:
You'll need to set a resistance value to use R/C mode.
Please see the "Multimeter" section here:
https://github.com/EspoTek/Labrador/wiki

Cheers,
~Chris

closed 2022-04-27 by EspoTek