espotek-org/Labrador #217: Single shot trigger not working on windows 10

lucasmolina255, 2022-04-03:
The single shot trigger does not work for any channel.  I start with a constant 5V reading. When I set the triggering value to 3V level on falling edge and check the "single shot" box, the readings become frozen.  Sending 0V to the monitored channel doesn't do anything.

[image]

EspoTek, 2022-04-04:
In the screenshot, it's looking for a rising edge.  Everything you've described is expected behaviour in this case.

It won't be able to require any rising edge at all from a 5V-0V transition.

closed 2022-04-04 by EspoTek

lucasmolina255, 2022-04-04:
Hi, thanks for replying so fast. Actually I tried both ways, falling edge - 5v initial and rising edge - 0v initial,  without any good response from the single shot triggering. In the screenshot I sent you I just had the scope set for the rising edge and forgot to switch it back but it wouldn't have changed anything.  I can record a full video if you want.