espotek-org/Labrador #222: Logic analyzer logging

david0rk, 2022-05-19:
for testing/reverse engineering purposes it would be nice to have an option to automatically log the serial decoder output to a logfile

EspoTek, 2022-05-20:
You know how sometimes people say something that is so plainly obvious that you wonder how you didn't think of it yourself?  This is one of those things.

Why have I not done this?

david0rk, 2022-05-20:
i realize you can just copy/paste from the text box, but logging to file in plaintext or hex dump would be ideal.

i grabbed the code and looked around during my break, it looks like the magic is happening in uartstyledecoder.cpp

EspoTek, 2022-05-20:
It is, and knowing me there'd be a sprintf() call too.  Literally as simple as replacing that with a fprintf()!

mi-hol, 2025-02-10:
@david0rk is this still an open issue?
In case not, please close.

closed 2025-02-11 by david0rk