espotek-org/Labrador #226: Device will only measure 3 volts

JvilleJTW, 2022-08-13:
Trying to test my Labrador. Comparing it against my DMM on a simple circuit with an LED and resistor driven by a lab supply of 9 volts DC. DMM correctly shows 7.6 volts across resistor. Labrador in Multimeter mode shows 3V consistently. Using CH1, CH2 and Gnd. terminals hooked up to test leads. I have tried increasing voltage headroom, but this appears to only affect the onboard voltage supply. Neither DC scope channel makes any sense either. Using a Mac laptop.

EspoTek, 2022-08-15:
Hi mate.  This isn't so much as a bug as questionable UI design.  Zoom out the scope trace so that the analog waveform is visible at 7.6V, and the digital display should start updating with the correct values too.

JvilleJTW, 2022-08-16:
Chris--

This makes no sense. Neither the multimeter nor the scope are reading the correct figures. It's not close; they are off by 4-5 volts. I am abandoning this project.

JTW

closed 2022-08-17 by EspoTek