espotek-org/Labrador #227: Connecting to Windows 10 and Mac OS 11 causes Labrador to crash in background

akaBryanB, 2022-08-14:
Device was working fine on both my Mac and PC. A couple of days ago the device would not connect on both computers. The application did not exhibit any visible problem, just that the hardware could not connect. The power light was lit on the board when plugged in, and I've tried several different USB cables. The USB connection was direct - there was no USB Hub.
I checked the logs on both computers, and found that the app was crashing in the background. The device will not go into bootloader mode, either by using the app, or shorting Digital Out 1 to GND.

I'm really baffled. Maybe the firmware is corrupted?

I've pasted only the header of both logs and attached both log files because the Mac log is 32000+ lines.

--------------------------------------------------
### Start of Windows Log
--------------------------------------------------
Log Name:      Application
Source:        Application Error
Date:          2022-08-12 2:43:56 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      Rubin
Description:
Faulting application name: Labrador.exe, version: 0.0.0.0, time stamp: 0x6268e050
Faulting module name: ntdll.dll, version: 10.0.19041.1806, time stamp: 0x1000a5b9
Exception code: 0xc0000374
Fault offset: 0x00000000000ff609
Faulting process id: 0x6bc
Faulting application start time: 0x01d8ae8c22a1fcaa
Faulting application path: C:\Program Files\EspoTek\EspoTek Labrador\Labrador.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 103be389-65bd-46b3-b6f2-62937c9d8331
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
\<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event"\>
  \<System>
    \<Provider Name="Application Error" />
    \<EventID Qualifiers="0">1000\</EventID>
    \<Version>0\</Version>
    \<Level>2\</Level>
    \<Task>100\</Task>
    \<Opcode>0\</Opcode>
    \<Keywords>0x80000000000000\</Keywords>
    \<TimeCreated SystemTime="2022-08-12T20:43:56.9773512Z" />
    \<EventRecordID>5095\</EventRecordID>
    \<Correlation />
    \<Execution ProcessID="0" ThreadID="0" />
    \<Channel>Application\</Channel>
    \<Computer>Rubin\</Computer>
    \<Security />
  \</System>
  \<EventData>
    \<Data>Labrador.exe\</Data>
    \<Data>0.0.0.0\</Data>
    \<Data>6268e050\</Data>
    \<Data>ntdll.dll\</Data>
    \<Data>10.0.19041.1806\</Data>
    \<Data>1000a5b9\</Data>
    \<Data>c0000374\</Data>
    \<Data>00000000000ff609\</Data>
    \<Data>6bc\</Data>
    \<Data>01d8ae8c22a1fcaa\</Data>
    \<Data>C:\Program Files\EspoTek\EspoTek Labrador\Labrador.exe\</Data>
    \<Data>C:\WINDOWS\SYSTEM32\ntdll.dll\</Data>
    \<Data>103be389-65bd-46b3-b6f2-62937c9d8331\</Data>
    \<Data>
    \</Data>
    \<Data>
    \</Data>
  \</EventData>
\</Event>

--------------------------------------------------
### Start of Mac OS log header
--------------------------------------------------
Process:               Labrador [61169]
Path:                  /Applications/Labrador.app/Contents/MacOS/Labrador
Identifier:            com.EspoTek.Labrador
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Labrador [61169]
User ID:               502

Date/Time:             2022-08-12 18:23:44.390 -0600
OS Version:            macOS 11.6.5 (20G527)
Report Version:        12
Anonymous UUID:        04D7036E-E042-86B6-3469-01709CD988B8

Time Awake Since Boot: 130000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [61169]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtWidgets      	0x000000010091fcc8 0x1008fb000 + 150728
1   org.qt-project.QtWidgets      	0x0000000100921e27 QHBoxLayout::~QHBoxLayout() + 39
2   org.qt-project.QtWidgets      	0x0000000100936115 QWidget::~QWidget() + 517
3   org.qt-project.QtWidgets      	0x0000000100936bbe QWidget::~QWidget() + 14
4   org.qt-project.QtCore         	0x00000001016c52d9 QObjectPrivate::deleteChildren() + 281
5   org.qt-project.QtWidgets      	0x000000010093637e QWidget::~QWidget() + 1134
6   org.qt-project.QtWidgets      	0x0000000100936bbe QWidget::~QWidget() + 1
7   org.qt-project.QtCore         	0x00000001016c52d9 QObjectPrivate::deleteChildren() + 281
8   org.qt-project.QtWidgets      	0x000000010093637e QWidget::~QWidget() + 1134
9   com.EspoTek.Labrador          	0x0000000100682e91 main + 81
10  libdyld.dylib                 	0x00007fff2062bf3d start + 1

[Crash Log Windows 10.txt](https://github.com/EspoTek/Labrador/files/9333986/Crash.Log.Windows.10.txt)
[Crash Log Mac OS 11.txt](https://github.com/EspoTek/Labrador/files/9333987/Crash.Log.Mac.OS.11.txt)

EspoTek, 2022-08-15:
This is odd.

Can you please provide a dump of the output from the lsusb command on macOS?
I'll need two dumps, one with the board connected and one without.
From there, we'll be able to see whether or not the board is detected at all, and if so what mode it's in.

If you don't have lsusb installed, you can install it via homebrew:
`/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"` (this command installs homebrew)
`brew install lsusb`

Then you can run it from the terminal by simply entering `lsusb`

akaBryanB, 2022-08-15:
Here you go.

--Bryan

[no_board_connected.txt](https://github.com/EspoTek/Labrador/files/9338657/no_board_connected.txt)
[board_connected.txt](https://github.com/EspoTek/Labrador/files/9338659/board_connected.txt)

EspoTek, 2022-08-16:
That's not good - looks like they're the same output.
This means the device isn't detected at all.

Are you absolutely sure the USB cable you're using has data lines?  You'll know this is the case if you can connect a device such as an Android phone, if makes the "USB Device Connected" noise (on Windows), and lets you transfer data to/from it.

akaBryanB, 2022-08-16:
I think I found the problem. I believe that the USB 2 port I was using did not have enough power. 
I plugged in a Raspberry Pi Pico to test the cable, and the Pico would not connect. Then I remembered that I had been connecting the Pico to my USB 3.1 PCI card, which I am powering from a SATA port (DUH, dim bulb). When I plugged the Pico into the USB 3.1 port, the Pico connected. Then when I plugged in the Oscilloscope it worked.
I still need to figure out what is going on with the Windows computer, but now that I know that the board is OK, I should be able track it down.
Thanks for your help. I really appreciate it. Thanks for building such a cool device.

EspoTek, 2022-08-17:
Awesome, good to hear!

I'll close off the ticket now.  If there's something Windows specific, feel free to contact me at admin@espotek.com

Cheers,
~Chris

closed 2022-08-17 by EspoTek